axoloti/axoloti #259: SDCard mounter: request for testing

JohannesTaelman, 2015-11-18:
"Enter card reader mode" switches from normal operation to card reader mode. The sdcard can only be mounted once, so it is not possible to integrate card reader mode with normal operation. 

I only tested on Windows 7. I'm looking for confirmation that it works on Linux and MacOS.
Unmounting/ejecting the mounted volume should trigger a microcontroller reset into normal mode.

Transfer speed is quite slow, due to small and no overlapping transactions.

natcl, 2015-11-24:
I'd be glad to test on osx, any info on how to test dev builds ?

JohannesTaelman, 2015-11-24:
https://github.com/axoloti/axoloti/blob/master/doc/compiling%20from%20source%20on%20osx.html
should cover it

natcl, 2015-11-25:
Works on OS X 10.11
Tested writing/reading a file and deleting files.

JohannesTaelman, 2015-11-25:
Does ejecting/unmounting the sdcard (in OSX finder) cause a microcontroller reset so you can go back to regular patching/playing mode without (un)plugging cables?

natcl, 2015-11-25:
Yes, after the card is ejected I see the green/red leds alternate and I can then press connect without having to disconnect the cable.

JohannesTaelman, 2015-11-25:
great, thanks for testing!

loopmasta, 2015-12-02:
I can confirm that it works as described above. Axoloti-Patcher (git pull & build), MacBook Air (11-inch, Mid 2012) with OSX 10.11

closed 2016-02-12 by axoloti