axoloti/axoloti #265: spat/pan m object output

vormplus, 2015-12-23:
Just noticed that the spat/pan m object lets a signal through on the left channel when the output is set to the right channel (see screenshot). You can see it on the first scope object.

When the ctrl/dial b object is set to -64.00, the right channel displays a flat line, which is the expected output.

[image]

JohannesTaelman, 2015-12-23:
+64 is really +63.9999... internally
The scope object does not round towards zero, so extremely small signals
will show up with one single pixel amplitude.
You'd need a very high amount of gain to notice this audibly.
Obscuring very small signals in the scope, well, they would become visible
again in more complex scenario's.

I don't think it is really an issue, but I'm open to hear more opinions!

vormplus, 2015-12-31:
From an end-user viewpoint, I think a flat line would be better in this case.
For developers, small noisy signals can be handy to debug patches.

closed 2016-08-09 by axoloti