axoloti/axoloti #267: Compiling with OpenJDK: JavaFX needed?

rekado, 2015-12-30:
I just got my axoloti core boards and would like to try the software, but I cannot seem to build it from source.  It does build the classes when the `-do-compile` target is executed, but the `bundle` target fails with this error:

~~~
[taskdef] Could not load definitions from resource com/sun/javafx/tools/ant/antlib.xml. It could not be found.
~~~

Is this an optional task?

I have OpenJDK7 available, but it seems that the build file requires JavaFX, which does not come with any of the released OpenJDK versions so far.

(Note that I cannot use the prebuilt binaries as I'm using a somewhat experimental system, the Guix System Distribution (GuixSD), which does not follow the FHS.  That's why I'd really like to build this from sources.)

JohannesTaelman, 2015-12-30:
The bundle task is to create a .deb/.dmg/.msi installer. Not required if
you build from source.

rekado, 2015-12-30:
Thanks for the confirmation!

rekado, 2016-01-01:
FWIW: I managed to build the cross-compiler and all of Axoloti on GNU Guix, but for some reason I cannot get it to play any sound upon loading a patch.

I also get the message "Firmware CRC mismatch detected" regularly.  Here's a log snippet:

~~~
INFO: Firmware version: 1.0.0.1, crc=0x35F3FC9D, entrypoint=0x20011000
Jan 01, 2016 10:09:43 PM axoloti.MainFrame setFirmwareID
SEVERE: Firmware CRC mismatch! Please flash the firmware first! Hardware firmware CRC = 35F3FC9D <> Software CRC = 458B59BB
Jan 01, 2016 10:09:43 PM axoloti.USBBulkConnection connect
INFO: Authentic Axoloti Core
~~~

When I answer "Yes" to the question whether I want to upload the firmware it seems to write to the USB device successfully and then the board reboots (blinking LEDs).  Upon reconnection, however, I'm asked the same question again.

TheTechnobear, 2016-01-02:
(firmware mismatch will often result in no sound, so that is the underlying issue) 

is the CRC changing after the firmware has been uploaded?
does the board reboot successfully?  Ive seen examples where the write to flash fails, usually power cycling the axoloti and trying again 'fixes' the issue.

also try using 'recovery mode' as discussed in the user guide on the community forum, this is the best 'if all else fails'.
(note: once you have successfully done the firmware upgrade, you can revert to using the normal flashing procedure, hopefully that will start working)

also, these things are best (I think) discussed on the community forum. as they relate to general 'user issues' rather then bugs etc in the software, so may be helpful for other users if they stumble across similar issues, and there is little point in having 2 discussion points. (and very few axoloti users use github)

rekado, 2016-01-02:
@TheTechnobear I have started a topic on the forum: http://community.axoloti.com/t/stuck-in-dfu-mode/811

The CRC does not change after the firmware has been "uploaded".  I did try recovery mode, but the board has not returned to normal operation.

Let's continue the discussion there :)

(I hope to fix this soon as I'd like to help working on the software; I'm close to submitting a package for Axoloti and a native ARM cross compiler for GNU Guix, which would make it trivial to set up Axoloti on any GNU/Linux system.)

closed 2016-01-02 by rekado