axoloti/axoloti #288: Better UI for the Object Editor (Mac OS X)

vormplus, 2016-02-17:
There is a huge usability issue with the object editor on Mac OS X. Windows does a great job when you put the tabs of the JTabbedPane on the left side. On the Mac, it's a horrible experience.

Best usage for tabbed panes is if you have no more than 7 tabs, and align them at the top of the frame. Since there are 13 items now, it might be good to rethink the UI for the object editor. Would be good if something similar to the layout on Windows was implemented in a different way so it looks more or less the same on all platforms.

[image]

TheTechnobear, 2016-02-17:
actually, I kind of agree , but for a different reason... it urgently needs keyboard shortcuts to properly move around the panes.

we are not going to be able to reduce the number of tabs, if anything we will need more over time.

we could move to hortizonal tabs, this would be better, as we need a 'wide' dialog anyway for editors.
(its why i dont like the windows L&F, its taking to much real estate)

the alternatives I can think of are:
- icons for tab , could take up much less space - does netbeans support?
- view menu - quick and simple, but not immediately obvious.  (gives us keyboard shortcuts for free)
- combo box for views - easy,  more obvious to users, but not very standard UI.

personally, I think id go for a view menu... unless you know how to implement icons for the tabs, and someone is willing to design the icons
the view menu is easy to extend, and I think once you get used to is reasonably obvious... bare in mind the menu is now in use for saving..

one question I have though, is if how netbeans is going to handle the editing, as the tabs are used in the editor, but perhaps we can just hide them at runtime

p.s. dont touch it now, as Im adding the new add to library functions, so we will get merge conflicts :)

vormplus, 2016-02-17:
I wouldn't go for icons only. Without a label, icons will only be "mystery meat navigation", and confuse the users even more. A fixed-width list of items on the left side of the window may be the best option.

Don't worry, I won't touch anything. I mainly do UI/UX design, don't have a lot of experience with implementation ;)

TheTechnobear, 2016-02-17:
Icons are fine, as long as you have tool tips, that then give you the current text.

I actually dont mind the current mac implementations, I prefer it over the windows one, 
the windows one takes way to much horizontal space 

anyway, seeing as I was there... I took a look
we cant change to using horizontal text, this is governed by the L&F, as its Apple UI standard to have it this way around, and I don't think its a good idea to change it. so I think we need another solution.

I did a quite look to see what it would look like with the tab on top, and actually I personally preferred it that way, you get bit more menu space.

the tabbed panes do support both icons and tooltips, so I think that would be my preferred option.

only one thing, I dont know what netbeans is like on windows, but Im having lots of trouble editing menus and these tabs in the GUI editor. some drop down menus dont appear in the editor any more, and I cannot scroll backwards and forward through the tabbed panes... its a nightmare changing these things!

bafonso, 2016-06-30:
With screens so big and high resolution is really the horizontal menu an issue? How do you move quickly between tabs in OSX? It makes for a slow workflow.

axoloti, 2016-08-20:
For traversing between tabs, you can use ctrl-pageup/ctrl-pagedown, on fruity keyboards this translates to ctrl-fn-up and ctrl-fn-down. I tried to activate ctrl-(shift)-tab for tab traversal but that did not work nicely.

closed 2016-08-20 by axoloti