axoloti/axoloti #362: Embed all subpatches

logsol, 2016-08-14:
I'd like to see a feature where you can simply collect and save all subpatches into the current patch for easier patch sharing. Similar to Abletons collect and save method, where all used samples are being copied into the project folder.

TheTechnobear, 2016-08-14:
yeah, I think this would be cool...

I wonder what the criteria should be though?  e.g. do we assume all factory patches are available to all users, therefore not include... but can the same not be said of all community sub-patches?

... also i think for it to work, with more using custom objects, we will need to also embed objects too.
(again, anything not in factory?!)

a nice 'side-effect' of this, would mean you could take an existing patch, collect it, and then use this for experimentation/extension. 

e.g. some users are playing with the demo patches, it would be nice for them to be able to 'collect' these, and then be able to modify their own copy .... oops, this means they might want the factory sub patches 'collected'

perhaps the best way, is a kind of dialog, where you can decide what is 'collected' (embedded)  i.e. you can place a tick box, against  each library you want to collect, and specify objects and/or patches.

logsol, 2016-08-15:
Originally I only thought of those sub patches that need to be in the same folder as the main patch, but you're also making a point. Ableton actually does it similarly.

[image]