axoloti/axoloti #369: Create an osx package to avoid using discrete GPU

bafonso, 2016-09-03:
Hi,

Apparently you can avoid java GUI interface using the discrete GPU if you create an OSX package and enter the right information in the Info.plist. See [here](https://developer.apple.com/library/mac/qa/qa1734/_index.html)

TheTechnobear, 2016-09-03:
as far as i can tell this is currently not possible using javaFX.
theres is an open enhancement against javaFX for it, 
but for now  there appears to be no way to add additional entries to the info.plist in the ant task

also we are not explicitly using openGL currently, so not sure if this would make much of a different given the tech note you link to.
but its worth bearing in mind if the rendering is changed, which may be possible after the MVC work is completed.

bafonso, 2016-09-03:
I thought it was fixed, see [here](https://bugs.openjdk.java.net/browse/JDK-8132775). Maybe I don't understand how it works to have it merged into the code we're using. You can't use ant task only, you need to use a .jar that [creates the package](http://docs.oracle.com/javase/7/docs/technotes/guides/jweb/packagingAppsForMac.html).

TheTechnobear, 2016-09-03:
is this relevant?
its against the openJDK, not the official java shipped by apple, which we use.... do we know this also had the same bug?
even if we used it, its just a bug that was fixed.. it makes no mention that you need to use the this OS X flag to activate it.

there are different ways to create an app from a package... I chose javaFX since it seemed to be the most 'up to date way' and had ant integration, was cross-platform the I looked at it... 
(btw, when I packaged axoloti, I did briefly look at the appbundler-1.0.jar method too, but cant remember if there was another reason i decided against it)

if you have experience using the appbundler-1.0.jar, and think its better. Id be interested to hear more, and if you submit a PR, that would be interesting to see. I don't think we are 'stuck on javaFX' as we only use it as a packaged and app integrator.

anyway, I'm not sure id do this, as perhaps this is not the underlying issue.

@JohannesTaelman  was talking about this recently, as he was looking at graphics performance.
so I'm sure he will pop along shortly, and add some details.

axoloti, 2016-09-03:
Looks pretty specific for Mac's that have both an integrated and a discrete GPU. My 2015 13" MBP only has an integrated GPU, I will not be able to test this. And forgive me, I don't have experience with an integrated+discrete gpu system, I don't really understand the impact. What is the benefit you expect from preventing to switch to discrete graphics? Apparently gfxCardStatus can force-disable the discrete graphics, can you verify this is effectively beneficial somehow?

TheTechnobear, 2016-09-03:
as far as I can tell, basically this switches the preference to using integrated graphics over discrete.
this saves on energy consumption (and so battery life), but presumably at the expense of graphics performance.

you can test it out using:
[code]
defaults write /Applications/Axoloti.app/Contents/Info.plist NSSupportsAutomaticGraphicsSwitching -bool YES
[/code]

a quick test on my MBP, didnt really show any appreciable difference according to the energy consumption figures in Activity Monitor.  (which also shows if this is active or not, labelled as 'requires high performance GPU)

TheTechnobear, 2016-09-03:
Ive found a way to be able to specify the Info.plist in javafx, but frankly its a bit of a pain to do.
so Id only do it, if we can get a bit more proof that its actually going to be worth it for anyone.

bafonso, 2016-09-03:
If you use gfxCardStatus, this is a moot point. Using the discretes shortens your average battery life roughly 1-1h30 and more likely to trigger fan. Whether that's relevant, it is up to each person.

In my mind there is not reason why the program should use given it has no use for it but I understand closing this issue if it's a pain to implement.

arisona, 2018-05-22:
The JRE-side issues with current JRE (8u72 and upwards) seem to be fixed.

Thus it is sufficient to add NSSupportsAutomaticGraphicsSwitching with "true" to Axoloti.app's Info.plist.

See Apple documentation: https://developer.apple.com/library/content/qa/qa1734/_index.html

See JRE bug fix: https://bugs.openjdk.java.net/browse/JDK-8132775

I've tested it with the current Axoloti release, looks fine - macOS remains on the integrated GPU when the app is running. 

I'd like to open a pull request, however: How is Axoloti.app built? I can't find any references to Info.plist in the git repo.

nicolasmiller, 2018-05-22:
Take a look at the section starting on line 356 of build.xml. This is where JavaFX is used to output a Mac-specific bundle. I'm guessing that what we're doing produces whatever default Info.plist that JavaFX knows how to produce for a Mac bundle. There is probably some way to add additional overrides.

arisona, 2018-05-22:
Nice - I'll check!

arisona, 2018-05-22:
Okay, not as easy as I thought: fx:deploy doesn't seem to support custom entries for Info.plist. It seems possibly to provide an entire Info.plist which is manually maintained, which I don't really like.

One workaround would be to mount the .dmg after fx:deploy, patch Info.plist and then recreate a new .dmg -- a bit awkward, but not too hard to implement (all necessary command line tools should be available on a default macOS installation).

JohannesTaelman, 2018-05-22:
Maybe adding an entire Info.plist is not too bad?

arisona, 2018-05-23:
Had a brief look at Info.plist - except for the Axoloti version (which can be set using "defaults" command), there are no changing items. So maybe it's indeed not too bad.