axoloti/axoloti #372: Axoloti patcher crashes randomly in yosemite

servandisco, 2016-10-22:
Process:               Axoloti [1173]
Path:                  /Applications/Axoloti.app/Contents/MacOS/Axoloti
Identifier:            axoloti
Version:               1.0.11 (1.0.11)
Code Type:             X86-64 (Native)
Parent Process:        ??? [1]
Responsible:           Axoloti [1173]
User ID:               501

Date/Time:             2016-10-22 20:14:37.132 +0200
OS Version:            Mac OS X 10.10.5 (14F27)
Report Version:        11
Anonymous UUID:        051FAA25-AF91-A223-861C-18D18CF3A0FC

Sleep/Wake UUID:       19CBF9D4-C743-4B61-9A74-F3642E8C0C41

Time Awake Since Boot: 35000 seconds
Time Since Wake:       18000 seconds

Crashed Thread:        19  Java: AWT-EventQueue-0

Exception Type:        EXC_BAD_INSTRUCTION (SIGABRT)
Exception Codes:       0x0000000000000001, 0x0000000000000000

Application Specific Information:
objc[1173]: Cannot form weak reference to instance (0x608000360240) of class _CSIRenditionBlockData. It is possible that this object was over-released, or is in the process of deallocation.
abort() called

Thread 0:: AppKit Thread  Dispatch queue: com.apple.main-thread
0   libsystem_kernel.dylib        	0x00007fff829995da syscall_thread_switch + 10
1   libsystem_platform.dylib      	0x00007fff8640dce0 _os_lock_handoff_lock_slow + 98
2   libobjc.A.dylib               	0x00007fff88582906 objc_object::sidetable_clearDeallocating() + 64
3   libobjc.A.dylib               	0x00007fff88568651 objc_destructInstance + 145
4   libobjc.A.dylib               	0x00007fff88568595 object_dispose + 22
5   com.apple.coreui              	0x00007fff8993d750 -[_CSIRenditionBlockData dealloc] + 56
6   libobjc.A.dylib               	0x00007fff8858289c objc_object::sidetable_release(bool) + 236
7   libdispatch.dylib             	0x00007fff882dd323 _dispatch_call_block_and_release + 12
8   libdispatch.dylib             	0x00007fff882d8c13 _dispatch_client_callout + 8
9   libdispatch.dylib             	0x00007fff882e4cbf _dispatch_main_queue_callback_4CF + 861
10  com.apple.CoreFoundation      	0x00007fff825c03f9 __CFRUNLOOP_IS_SERVICING_THE_MAIN_DISPATCH_QUEUE__ + 9
11  com.apple.CoreFoundation      	0x00007fff8257b68f __CFRunLoopRun + 2159
12  com.apple.CoreFoundation      	0x00007fff8257abd8 CFRunLoopRunSpecific + 296
13  com.apple.HIToolbox           	0x00007fff8ba7f56f RunCurrentEventLoopInMode + 235
14  com.apple.HIToolbox           	0x00007fff8ba7f2ea ReceiveNextEventCommon + 431
15  com.apple.HIToolbox           	0x00007fff8ba7f12b _BlockUntilNextEventMatchingListInModeWithFilter + 71
16  com.apple.AppKit              	0x00007fff8a36a8ab _DPSNextEvent + 978
17  com.apple.AppKit              	0x00007fff8a369e58 -[NSApplication nextEventMatchingMask:untilDate:inMode:dequeue:] + 346
18  libosxapp.dylib               	0x00000001225e13aa -[NSApplicationAWT nextEventMatchingMask:untilDate:inMode:dequeue:] + 124
19  com.apple.AppKit              	0x00007fff8a35faf3 -[NSApplication run] + 594
20  libosxapp.dylib               	0x00000001225e114d +[NSApplicationAWT runAWTLoopWithApp:] + 156
21  libawt_lwawt.dylib            	0x000000012256e4b3 -[AWTStarter starter:] + 905
22  com.apple.Foundation          	0x00007fff8530cdd0 __NSThreadPerformPerform + 293
23  com.apple.CoreFoundation      	0x00007fff82589a01 __CFRUNLOOP_IS_CALLING_OUT_TO_A_SOURCE0_PERFORM_FUNCTION__ + 17
24  com.apple.CoreFoundation      	0x00007fff8257bb8d __CFRunLoopDoSources0 + 269
25  com.apple.CoreFoundation      	0x00007fff8257b1bf __CFRunLoopRun + 927
26  com.apple.CoreFoundation      	0x00007fff8257abd8 CFRunLoopRunSpecific + 296
27  libjli.dylib                  	0x000000010c96c8fd CreateExecutionEnvironment + 871
28  libjli.dylib                  	0x000000010c968624 JLI_Launch + 1952
29  libpackager.dylib             	0x000000010af544c7 JavaLibrary::JavaVMCreate(unsigned long, char**) + 183
30  libpackager.dylib             	0x000000010af51db1 JavaVirtualMachine::StartJVM() + 3457
31  libpackager.dylib             	0x000000010af50f60 RunVM() + 16
32  libpackager.dylib             	0x000000010af5e81f start_launcher + 1791
33  axoloti                       	0x000000010aec5cd9 main + 217
34  axoloti                       	0x000000010aec5bf4 start + 52

Thread 1:: Dispatch queue: com.apple.libdispatch-manager
0   libsystem_kernel.dylib        	0x00007fff8299f232 kevent64 + 10
1   libdispatch.dylib             	0x00007fff882dba6a _dispatch_mgr_thread + 52

Thread 2:
[... 656 lines cut ...]
TOTAL, minus reserved VM space      4.8G
 

Model: MacBookAir5,1, BootROM MBA51.00EF.B03, 2 processors, Intel Core i5, 1.7 GHz, 4 GB, SMC 2.4f19
Graphics: Intel HD Graphics 4000, Intel HD Graphics 4000, Built-In
Memory Module: BANK 0/DIMM0, 2 GB, DDR3, 1600 MHz, 0x80AD, 0x484D54333235533643465238412D50422020
Memory Module: BANK 1/DIMM0, 2 GB, DDR3, 1600 MHz, 0x80AD, 0x484D54333235533643465238412D50422020
AirPort: spairport_wireless_card_type_airport_extreme (0x14E4, 0xE9), Broadcom BCM43xx 1.0 (7.15.166.24.3)
Bluetooth: Version 4.3.6f3 16238, 3 services, 18 devices, 1 incoming serial ports
Network Service: Wi-Fi, AirPort, en0
Network Service: 10.10, Ethernet, en3
Serial ATA Device: APPLE SSD TS128E, 121.33 GB
USB Device: Axoloti Core
USB Device: Hub
USB Device: FaceTime HD Camera (Built-in)
USB Device: Hub
USB Device: USB 2.0 Hub
USB Device: USB 2.0 Hub
USB Device: Komplete Audio 6
USB Device: Mad Catz R.A.T.3 Mouse
USB Device: Hub
USB Device: Apple Internal Keyboard / Trackpad
USB Device: BRCM20702 Hub
USB Device: Bluetooth USB Host Controller
Thunderbolt Bus: MacBook Air, Apple Inc., 23.4

servandisco, 2016-10-23:
again:

```
Process:               Axoloti [2775]
Path:                  /Applications/Axoloti.app/Contents/MacOS/Axoloti
Identifier:            axoloti
Version:               1.0.11 (1.0.11)
Code Type:             X86-64 (Native)
Parent Process:        ??? [1]
Responsible:           Axoloti [2775]
User ID:               501

Date/Time:             2016-10-23 19:52:36.853 +0200
OS Version:            Mac OS X 10.10.5 (14F27)
Report Version:        11
Anonymous UUID:        051FAA25-AF91-A223-861C-18D18CF3A0FC

Sleep/Wake UUID:       9728987C-0A3E-4891-A26F-E15842E4236A

Time Awake Since Boot: 67000 seconds
Time Since Wake:       8300 seconds

Crashed Thread:        19  Java: AWT-EventQueue-0

Exception Type:        EXC_BAD_ACCESS (SIGABRT)
Exception Codes:       EXC_I386_GPFLT

Application Specific Information:
abort() called

Thread 0:: AppKit Thread  Dispatch queue: com.apple.main-thread
0   libsystem_kernel.dylib        	0x00007fff829994de mach_msg_trap + 10
1   libsystem_kernel.dylib        	0x00007fff8299864f mach_msg + 55
2   com.apple.CoreFoundation      	0x00007fff8257beb4 __CFRunLoopServiceMachPort + 212
3   com.apple.CoreFoundation      	0x00007fff8257b37b __CFRunLoopRun + 1371
4   com.apple.CoreFoundation      	0x00007fff8257abd8 CFRunLoopRunSpecific + 296
5   com.apple.HIToolbox           	0x00007fff8ba7f56f RunCurrentEventLoopInMode + 235
6   com.apple.HIToolbox           	0x00007fff8ba7f2ea ReceiveNextEventCommon + 431
7   com.apple.HIToolbox           	0x00007fff8ba7f12b _BlockUntilNextEventMatchingListInModeWithFilter + 71
8   com.apple.AppKit              	0x00007fff8a36a8ab _DPSNextEvent + 978
9   com.apple.AppKit              	0x00007fff8a369e58 -[NSApplication nextEventMatchingMask:untilDate:inMode:dequeue:] + 346
10  libosxapp.dylib               	0x0000000117d903aa -[NSApplicationAWT nextEventMatchingMask:untilDate:inMode:dequeue:] + 124
11  com.apple.AppKit              	0x00007fff8a35faf3 -[NSApplication run] + 594
12  libosxapp.dylib               	0x0000000117d9014d +[NSApplicationAWT runAWTLoopWithApp:] + 156
13  libawt_lwawt.dylib            	0x000000011841f4b3 -[AWTStarter starter:] + 905
14  com.apple.Foundation          	0x00007fff8530cdd0 __NSThreadPerformPerform + 293
15  com.apple.CoreFoundation      	0x00007fff82589a01 __CFRUNLOOP_IS_CALLING_OUT_TO_A_SOURCE0_PERFORM_FUNCTION__ + 17
16  com.apple.CoreFoundation      	0x00007fff8257bb8d __CFRunLoopDoSources0 + 269
17  com.apple.CoreFoundation      	0x00007fff8257b1bf __CFRunLoopRun + 927
18  com.apple.CoreFoundation      	0x00007fff8257abd8 CFRunLoopRunSpecific + 296
19  libjli.dylib                  	0x00000001028e48fd CreateExecutionEnvironment + 871
20  libjli.dylib                  	0x00000001028e0624 JLI_Launch + 1952
21  libpackager.dylib             	0x0000000100ecc4c7 JavaLibrary::JavaVMCreate(unsigned long, char**) + 183
22  libpackager.dylib             	0x0000000100ec9db1 JavaVirtualMachine::StartJVM() + 3457
23  libpackager.dylib             	0x0000000100ec8f60 RunVM() + 16
24  libpackager.dylib             	0x0000000100ed681f start_launcher + 1791
25  axoloti                       	0x0000000100e3dcd9 main + 217
26  axoloti                       	0x0000000100e3dbf4 start + 52

Thread 1:: Dispatch queue: com.apple.libdispatch-manager
0   libsystem_kernel.dylib        	0x00007fff8299f232 kevent64 + 10
1   libdispatch.dylib             	0x00007fff882dba6a _dispatch_mgr_thread + 52

Thread 2:
0   libsystem_kernel.dylib        	0x00007fff8299e48a __semwait_signal + 10
1   libjli.dylib                  	0x00000001028e3eba ContinueInNewThread0 + 102
2   libjli.dylib                  	0x00000001028e13eb ContinueInNewThread + 201
3   libjli.dylib                  	0x00000001028e3c8f JVMInit + 315
4   libjli.dylib                  	0x00000001028e1169 JLI_Launch + 4837
5   libpackager.dylib             	0x0000000100ecc4c7 JavaLibrary::JavaVMCreate(unsigned long, char**) + 183
[... 697 lines cut ...]

Model: MacBookAir5,1, BootROM MBA51.00EF.B03, 2 processors, Intel Core i5, 1.7 GHz, 4 GB, SMC 2.4f19
Graphics: Intel HD Graphics 4000, Intel HD Graphics 4000, Built-In
Memory Module: BANK 0/DIMM0, 2 GB, DDR3, 1600 MHz, 0x80AD, 0x484D54333235533643465238412D50422020
Memory Module: BANK 1/DIMM0, 2 GB, DDR3, 1600 MHz, 0x80AD, 0x484D54333235533643465238412D50422020
AirPort: spairport_wireless_card_type_airport_extreme (0x14E4, 0xE9), Broadcom BCM43xx 1.0 (7.15.166.24.3)
Bluetooth: Version 4.3.6f3 16238, 3 services, 18 devices, 1 incoming serial ports
Network Service: Wi-Fi, AirPort, en0
Network Service: 10.10, Ethernet, en3
Serial ATA Device: APPLE SSD TS128E, 121.33 GB
USB Device: Axoloti Core
USB Device: Hub
USB Device: FaceTime HD Camera (Built-in)
USB Device: Hub
USB Device: USB 2.0 Hub
USB Device: LPD8
USB Device: USB 2.0 Hub
USB Device: Komplete Audio 6
USB Device: Mad Catz R.A.T.3 Mouse
USB Device: Hub
USB Device: Apple Internal Keyboard / Trackpad
USB Device: BRCM20702 Hub
USB Device: Bluetooth USB Host Controller
Thunderbolt Bus: MacBook Air, Apple Inc., 23.4
```

servandisco, 2016-10-24:
Process:               Axoloti [4752]
Path:                  /Applications/Axoloti.app/Contents/MacOS/Axoloti
Identifier:            axoloti
Version:               1.0.11 (1.0.11)
Code Type:             X86-64 (Native)
Parent Process:        ??? [1]
Responsible:           Axoloti [4752]
User ID:               501

Date/Time:             2016-10-24 02:37:22.429 +0200
OS Version:            Mac OS X 10.10.5 (14F27)
Report Version:        11
Anonymous UUID:        051FAA25-AF91-A223-861C-18D18CF3A0FC

Sleep/Wake UUID:       9728987C-0A3E-4891-A26F-E15842E4236A

Time Awake Since Boot: 92000 seconds
Time Since Wake:       32000 seconds

Crashed Thread:        19  Java: AWT-EventQueue-0

Exception Type:        EXC_BAD_ACCESS (SIGABRT)
Exception Codes:       EXC_I386_GPFLT

Application Specific Information:
abort() called

Thread 0:: AppKit Thread  Dispatch queue: com.apple.main-thread
0   libsystem_kernel.dylib        	0x00007fff829995da syscall_thread_switch + 10
1   libsystem_platform.dylib      	0x00007fff8640dce0 _os_lock_handoff_lock_slow + 98
2   libobjc.A.dylib               	0x00007fff88582906 objc_object::sidetable_clearDeallocating() + 64
3   libobjc.A.dylib               	0x00007fff88568651 objc_destructInstance + 145
4   com.apple.CoreFoundation      	0x00007fff82522089 CFRelease + 1033
5   com.apple.CoreFoundation      	0x00007fff82533b2b parseXMLElement + 7371
6   com.apple.CoreFoundation      	0x00007fff82534575 getContentObject + 741
7   com.apple.CoreFoundation      	0x00007fff82532dee parseXMLElement + 3982
8   com.apple.CoreFoundation      	0x00007fff82534575 getContentObject + 741
9   com.apple.CoreFoundation      	0x00007fff825336ab parseXMLElement + 6219
10  com.apple.CoreFoundation      	0x00007fff82534575 getContentObject + 741
11  com.apple.CoreFoundation      	0x00007fff82532324 parseXMLElement + 1220
12  com.apple.CoreFoundation      	0x00007fff8253103f _CFPropertyListCreateFromUTF8Data + 2591
13  com.apple.CoreFoundation      	0x00007fff8252fcce _CFPropertyListCreateWithData + 1054
14  com.apple.CoreFoundation      	0x00007fff8252f70d CFPropertyListCreateWithData + 93
15  com.apple.CoreGraphics        	0x00007fff8bd5d0d9 CGSPropertyListCreateWithSerializedData + 44
16  com.apple.CoreGraphics        	0x00007fff8be5f116 createPlistFromData + 72
17  com.apple.CoreGraphics        	0x00007fff8be5f0c3 CGWindowListCopyWindowInfo + 100
18  libawt_lwawt.dylib            	0x0000000124e65ecf +[AWTWindow getTopmostWindowUnderMouseID] + 126
19  libawt_lwawt.dylib            	0x0000000124e65de5 +[AWTWindow getTopmostWindowUnderMouse] + 78
20  libawt_lwawt.dylib            	0x0000000124e68e76 Java_sun_lwawt_macosx_CPlatformWindow_nativeGetTopmostPlatformWindowUnderMouse + 109
21  ???                           	0x000000011063726c 0 + 4569920108
22  ???                           	0x0000000110dc2820 0 + 4577830944

Thread 1:: Dispatch queue: com.apple.libdispatch-manager
0   libsystem_kernel.dylib        	0x00007fff8299f232 kevent64 + 10
1   libdispatch.dylib             	0x00007fff882dba6a _dispatch_mgr_thread + 52

Thread 2:
0   libsystem_kernel.dylib        	0x00007fff8299e48a __semwait_signal + 10
1   libjli.dylib                  	0x000000010f2f4eba ContinueInNewThread0 + 102
2   libjli.dylib                  	0x000000010f2f23eb ContinueInNewThread + 201
3   libjli.dylib                  	0x000000010f2f4c8f JVMInit + 315
4   libjli.dylib                  	0x000000010f2f2169 JLI_Launch + 4837
5   libpackager.dylib             	0x000000010d8dd4c7 JavaLibrary::JavaVMCreate(unsigned long, char**) + 183
6   libpackager.dylib             	0x000000010d8dadb1 JavaVirtualMachine::StartJVM() + 3457
7   libpackager.dylib             	0x000000010d8d9f60 RunVM() + 16
8   libpackager.dylib             	0x000000010d8e781f start_launcher + 1791
9   axoloti                       	0x000000010d84ecd9 main + 217
10  libjli.dylib                  	0x000000010f2f558f apple_main + 92
11  libsystem_pthread.dylib       	0x00007fff87dc105a _pthread_body + 131
12  libsystem_pthread.dylib       	0x00007fff87dc0fd7 _pthread_start + 176
[... 654 lines cut ...]
TOTAL, minus reserved VM space      4.2G
 

Model: MacBookAir5,1, BootROM MBA51.00EF.B03, 2 processors, Intel Core i5, 1.7 GHz, 4 GB, SMC 2.4f19
Graphics: Intel HD Graphics 4000, Intel HD Graphics 4000, Built-In
Memory Module: BANK 0/DIMM0, 2 GB, DDR3, 1600 MHz, 0x80AD, 0x484D54333235533643465238412D50422020
Memory Module: BANK 1/DIMM0, 2 GB, DDR3, 1600 MHz, 0x80AD, 0x484D54333235533643465238412D50422020
AirPort: spairport_wireless_card_type_airport_extreme (0x14E4, 0xE9), Broadcom BCM43xx 1.0 (7.15.166.24.3)
Bluetooth: Version 4.3.6f3 16238, 3 services, 18 devices, 1 incoming serial ports
Network Service: Wi-Fi, AirPort, en0
Network Service: 10.10, Ethernet, en3
Serial ATA Device: APPLE SSD TS128E, 121.33 GB
USB Device: Axoloti Core
USB Device: Hub
USB Device: FaceTime HD Camera (Built-in)
USB Device: Hub
USB Device: USB 2.0 Hub
USB Device: USB 2.0 Hub
USB Device: Komplete Audio 6
USB Device: Mad Catz R.A.T.3 Mouse
USB Device: Hub
USB Device: Apple Internal Keyboard / Trackpad
USB Device: BRCM20702 Hub
USB Device: Bluetooth USB Host Controller
Thunderbolt Bus: MacBook Air, Apple Inc., 23.4

TheTechnobear, 2017-05-10:
I dont have a mac with 10.10 to test on, is this still happening?
does it always happen or under certain scenarios?

generally this appears to be a java issue, with drawing something on to the screen. this is not axoloti code  as such, but rather an issue with the JVM... 
perhaps you can upgrade the version of java installed? 

note: I originally ran axoloti in 10.10, and then 10.11 and now 10.12 with no issues (also 10.7!) , so its not a general issues, perhaps the JVM and the MacBook air/gpu combination.

I'll close this now, as there really aren't enough details for reproducing, and no one else has raised. 
(forum has a number of posts with users using 10.10)

closed 2017-05-10 by TheTechnobear