axoloti/axoloti #378: Ordering of includes

niklasnisbeth, 2017-01-21:
Files in the <includes> section of a patch are not added to the top of the generated file in the order they are specified.

This is apparently because the relevant functions in Patch.java and AxoObject.java use HashSets, which do not preserve ordering. 

One can use Treesets, which preserve the insertion order but still have set semantics. There's a performance hit, but I assume it'll be impossible to even measure for the number of inserts you're going to see in a normal patch.

If I've not missed any concerns, I'll fix it myself and make a pull request.

deadsy, 2018-07-07:
Still broken.
Makes multiple include files useless if there are dependencies between them.

In *.axo
```
    <includes>
      <include>./a.h</include>
      <include>./b.h</include>
      <include>./c.h</include>
      <include>./d.h</include>
      <include>./e.h</include>
      <include>./f.h</include>
      <include>./g.h</include>
    </includes>
```

In xpatch.cpp
```
#include "/home/jasonh/work/axo/work/objects/sx1509/f.h"
#include "/home/jasonh/work/axo/work/objects/sx1509/g.h"
#include "/home/jasonh/work/axo/work/objects/sx1509/b.h"
#include "/home/jasonh/work/axo/work/objects/sx1509/c.h"
#include "/home/jasonh/work/axo/work/objects/sx1509/d.h"
#include "/home/jasonh/work/axo/work/objects/sx1509/e.h"
#include "/home/jasonh/work/axo/work/objects/sx1509/a.h"
```

TheTechnobear, 2019-12-20:
tested in 2.0.0, and is ok

closed 2019-12-20 by TheTechnobear