axoloti/axoloti #388: MPE - PB before note on ignored

TheTechnobear, 2017-03-06:
bug in generated code.

```
             } else if (msg == MIDI_PITCH_BEND) {
                    if (channel != 0
                            && (channel < lowChannel || channel > highChannel))
                        return;
                    int i;
                    for(i=0; i<3; i++) {
                        if (voiceChannel[i] == channel || channel == 0) {
                            getVoices()[i].MidiInHandler(dev, port, MIDI_PITCH_BEND + 0, data1, data2);
                        }
                    }
```

currently voiceChannel is set to 0xFF on note-off, so pitch bends / cc are ignored after note-off
this is incorrect, they still should be processed.

other things to check
a) sustain handling, double check the the above doesn't break sustain handling
b) voice allocation is done on NOTE on... 
this means when we very first start voiceChannel = 0xFF , so the very first note on a channel will set it.
which means the PB sent before the note on, still wont get sent.
c) does the voice get allocated to the same channel as the PB is on!

I think (c) is a big issue, and (b) is kind of a variation on it.

the 'obvious' solution is to implement this such that voice channel = (midi channel -1) 
i.e. fix the relationship between voices and channels.
this is nice because its simple and fast, you don't need any voice allocation, you just send your messages straight thru.

BUT... it has 2  limitations.
a) you cannot play multiple notes on the same midi channel.
this is usually only done if you run out of voices, and usually in MPE/expressive controllers you really dont want this, because suddenly you loose note independence.
BUT, Ive seen users using the MPE object for non MPE , I think probably because of the included NOTE+PB = pitch behaviour.
... these users would not like the change.
i suspect a replacement keyb object is the answer for them

b) you need to match midi channel count and voice count
this is kind of like (a) but subtle different.
currently, if I have a controller setup and with rotating midi channels 2-16, I can play with 4 voices, and it will work.
If we move to fixed voice allocation it wont, you would need to set your rotation to the same count.
again, in MPE usage, this is already common...

TheTechnobear, 2017-03-06:
ok, I dont want to inflict a and b :)

the solution Im going to go for is less invasive too.
keep the voice handling the same,  but Im going to store the last PB and CC74 message for each channel (based on midi channel NOT voice id), when on note on, after voice allocation, I will send the recorded PB/CC74 before the note on.

the pb/cc74 can be stored in 32bit, so its only cost 15x 32bit. which I think is a reasonable trade-off for the avoiding the limitations.

TheSlowGrowth, 2017-03-08:
Actually, I would prefer sticking straight to the MPE spec with the two possible configurations:
* Main channel 1, notes on channels 2 and up
* Main channel 16, notes on channels 15 and down

Right now (if I'm not mistaken) it is not possible to use the second configuration. Hence, it is not possible to play two independent sounds from one controller.

Users would select one of the two configurations from a dropdown menu. A second dropdown would allow to select how many channels are used by the controller. 
The number of voices in the subpatch can still be un-coupled from the number of channels that are used, allowing sounds with a long release time to proberly decay. To ensure that there are always enough voices to serve the number of midi channels, the patch would select the number of midi channels that it uses like this:

    numMidiCh = (numRequestedMidiCh < numVoices)?numRequestedMidiCh:numVoices;

A little display below the dropdown menu could help the user understand which channels are effectively in use. That display could be as simple as 16 checkboxes, one for each midi channel. Channels used by the subpatch would have a tick in the corresponding box. In the case that there are less sound enging voices than requested midi channels, a small message could be displayed, saying something like: 

I can see that this would not allow a user to simply setup the controller for notes on channels 2-16 and leave it like that, regardless of the sound generator that is used. I don't see a clever way to hide those protocol details from the user...
IMO it is weird, that the MPE spec doesn't define a way for the sound generator to tell the controller which channels to use.

TheTechnobear, 2017-03-08:
this is unrelated to the issue at hand, it would be a new feature request

BUT there is no point implementing it...
the split you mention is actually from an older MPE spec, the newer 1.25a spec which was sent to the MMA for approval has a completely different approach. 
1.25a proposes a much more elaborate (and complicated) approach allowing for multiple splits,unforuntately its incompatible with the ch 16 approach.
as its quite a bit of effort to implement, I decided to wait until its been approved, in case it changes again.

TheSlowGrowth, 2017-03-08:
Oh well, I wasn't aware of the 1.25a spec.
In that case it is indeed better to apply a quick hotfix to the issue and wait until the spec is finalized and more widely adopted