axoloti/axoloti #398: patch with windowPos x=0 opens then crashes when loaded into the patcher

DaveCasey, 2017-08-07:
I am not sure how my patch got to the point where I had a zero for that value, but once it happened, I was no longer able to load it into the patcher and work on that patch. (And I had grown rather fond of that patch).

To repro: create a new patch. Edit it in a text editor, and change the windowPos x value to zero.

`  
 <patch-1.0 appVersion="1.0.12">
   <nets/>
   <settings>
      <subpatchmode>no</subpatchmode>
   </settings>
   <notes><![CDATA[]]></notes>
   <windowPos>
      <x>0</x>
      <y>65</y>
      <width>400</width>
      <height>400</height>
   </windowPos>
</patch-1.0>
`

Workaround: Edit corrupt patch in a text editor and set x to 1.

[crash_repro.axp.zip](https://github.com/axoloti/axoloti/files/1205095/crash_repro.axp.zip)

JohannesTaelman, 2017-08-07:
Was the patch made on Windows, Mac or Linux?

DaveCasey, 2017-08-07:
Patch was created on Mac 10.12.6.