axoloti/axoloti #401: Testability

axoloti, 2017-09-27:
The current "test compilation" implementation in the gui does not give a good overview or allow easy diagnosis, it 'd be cool to move to generating a corresponding .cpp/.mk file for every .axp/.axh, perhaps without even opening the gui but as a cmd line option. That could in turn lead into travis-ifying compilation tests, triggered perhaps even by library commits.

axoloti, 2017-10-05:
Modules are missing, those are currently passed from GUI to make via environment variables.

option 1: export another file besides the .cpp file for every patch 
(possibly a makefile or a shellscript)

option 2: sneak modules info into a comment in the .cpp file, and parsed by the makefile

TODO: test on linux/windows

TheTechnobear, 2017-10-06:
I'd say, build all modules in a step before patch files are compiled ( good test anyway) 
This means your issue then is one one of linking.  

Option 1 then looks to be the most attractiive. 

What concerns me about all of this though, is this moves away from using the code in the patcher, which means it's possible to introduce issues which potentially compile in test, but fail in the patcher.