axoloti/axoloti #404: Unite runtime binaries and java app into one installer

axoloti, 2017-09-27:
The oracle/javafx bundling process removes executable flags from binaries, so it was (is) impossible to include working binaries. Maybe we can just move to a different packager than the Oracle FX bundler. Perhaps http://izpack.org/ ?

TheTechnobear, 2019-12-20:
is this relevant now?  (2.0.0) 
do you want to ship arm compiler , or leave as explicit step for user install?

rekado, 2019-12-20:
I've packaged the Axoloti patcher some years ago for the reproducible package manager GNU Guix.  Guix can create portable bundles with [`guix pack`](https://guix.gnu.org/manual/en/html_node/Invoking-guix-pack.html).  This would include the arm cross-compiler.  The downside is that it would only work for GNU/Linux systems (and possible macos via Docker, which is a format that `guix pack` can also generate).