axoloti/axoloti #424: GUI: Opening 'edit object definition' spawns duplicate on top of selected object

krizroring, 2018-03-03:
OS: Windows 10

When opening the 'edit object definition', creates a duplicatie of the selected object positioned on top of selected object. This new spawned object is a 'Ghost' object, not saved with the patch, but looks like a active object.

krizroring, 2018-03-04:
Futher investigation show this only happens when you open the 'edit object definition' by using right mouse button, not when using the small triangle in the upper left corner.

himijendrix24, 2018-06-27:
Same issue here.

The copy can't be deleted manually (select --> "del").

This happens on different machines with windows 7 and 10 machines.
I'm using build version 1.0.12-0-g080370d2.