axoloti/axoloti #430: Exception loop

wotwot, 2018-06-17:
I'm trying to run 7975cc8 on a fresh debian stretch 64bit install.     
After activating multiarch, installing and commenting the missing lib32* in build.sh, installing openjdk8 and compiling, axoloti.log contains          

```
defaulting to developer defaults, can be overridden
Axoloti Directories:
Current = /data/src/snd/axoloti/axoloti
Jar = /data/src/snd/axoloti/axoloti/dist
Release = /data/src/snd/axoloti/axoloti
Runtime = /data/src/snd/axoloti/axoloti
Firmware = /data/src/snd/axoloti/axoloti/firmware
AxolotiHome = /data/src/snd/axoloti/axoloti
Jun 17, 2018 3:26:54 PM axoloti.utils.Preferences SavePrefs
INFO: Saving preferences...
Jun 17, 2018 3:26:54 PM axoloti.utils.Preferences SavePrefs
INFO: preferences path : /data/src/snd/axoloti/axoloti/axoloti.prefs
Jun 17, 2018 3:26:54 PM axoloti.MainFrame$4 run
INFO: Axoloti version : 1.0.12-2-0-g7975cc87-dirty  build time : 17/06/2018 14:57:29 CEST
Jun 17, 2018 3:26:54 PM axoloti.MainFrame$3 publish
SEVERE: null
javax.swing.text.BadLocationException: Invalid insert
        at java.desktop/javax.swing.text.AbstractDocument.insertString(AbstractDocument.java:706)
        at axoloti.MainFrame$3.publish(MainFrame.java:214)
        at java.logging/java.util.logging.Logger.log(Logger.java:979)
        at java.logging/java.util.logging.Logger.doLog(Logger.java:1006)
        at java.logging/java.util.logging.Logger.log(Logger.java:1092)
        at axoloti.MainFrame$4.run(MainFrame.java:302)
        at java.desktop/java.awt.event.InvocationEvent.dispatch(InvocationEvent.java:313)
        at java.desktop/java.awt.EventQueue.dispatchEventImpl(EventQueue.java:764)
        at java.desktop/java.awt.EventQueue.access$500(EventQueue.java:97)
        at java.desktop/java.awt.EventQueue$3.run(EventQueue.java:717)
        at java.desktop/java.awt.EventQueue$3.run(EventQueue.java:711)
        at java.base/java.security.AccessController.doPrivileged(Native Method)
        at java.base/java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(ProtectionDomain.java:89)
        at java.desktop/java.awt.EventQueue.dispatchEvent(EventQueue.java:734)
        at java.desktop/java.awt.EventDispatchThread.pumpOneEventForFilters(EventDispatchThread.java:199)
        at java.desktop/java.awt.EventDispatchThread.pumpEventsForFilter(EventDispatchThread.java:124)
        at java.desktop/java.awt.EventDispatchThread.pumpEventsForHierarchy(EventDispatchThread.java:113)
        at java.desktop/java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:109)
        at java.desktop/java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:101)
        at java.desktop/java.awt.EventDispatchThread.run(EventDispatchThread.java:90)
```

followed by a couple of thousand lines in the style of      

```
Jun 17, 2018 3:26:54 PM axoloti.MainFrame$3 publish
SEVERE: null
javax.swing.text.BadLocationException: Invalid insert
        at java.desktop/javax.swing.text.AbstractDocument.insertString(AbstractDocument.java:706)
        at axoloti.MainFrame$3.publish(MainFrame.java:205)
        at java.logging/java.util.logging.Logger.log(Logger.java:979)
        at java.logging/java.util.logging.Logger.doLog(Logger.java:1006)
        at java.logging/java.util.logging.Logger.log(Logger.java:1117)
        at axoloti.MainFrame$3.publish(MainFrame.java:218)
        at java.logging/java.util.logging.Logger.log(Logger.java:979)
        at java.logging/java.util.logging.Logger.doLog(Logger.java:1006)
        at java.logging/java.util.logging.Logger.log(Logger.java:1092)
        at axoloti.MainFrame$4.run(MainFrame.java:302)
        at java.desktop/java.awt.event.InvocationEvent.dispatch(InvocationEvent.java:313)
        at java.desktop/java.awt.EventQueue.dispatchEventImpl(EventQueue.java:764)
        at java.desktop/java.awt.EventQueue.access$500(EventQueue.java:97)
        at java.desktop/java.awt.EventQueue$3.run(EventQueue.java:717)
        at java.desktop/java.awt.EventQueue$3.run(EventQueue.java:711)
        at java.base/java.security.AccessController.doPrivileged(Native Method)
        at java.base/java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(ProtectionDomain.java:89)
        at java.desktop/java.awt.EventQueue.dispatchEvent(EventQueue.java:734)
        at java.desktop/java.awt.EventDispatchThread.pumpOneEventForFilters(EventDispatchThread.java:199)
        at java.desktop/java.awt.EventDispatchThread.pumpEventsForFilter(EventDispatchThread.java:124)
        at java.desktop/java.awt.EventDispatchThread.pumpEventsForHierarchy(EventDispatchThread.java:113)
        at java.desktop/java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:109)
        at java.desktop/java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:101)
        at java.desktop/java.awt.EventDispatchThread.run(EventDispatchThread.java:90)

```
  
Is this install missing something ?

nicolasmiller, 2018-06-17:
This is related to https://github.com/axoloti/axoloti/pull/429 which was merged into experimental. There's a lame issue there related to how the Document API works that varies across JDK versions.

My guess is that possibly you are launching a Java runtime version other than 8. Are there multiple Javas on the system? You might need to explicitly choose 8.

Also, running the experimental branch might just work because it should be resilient to this issue now.

wotwot, 2018-06-17:
Yes, jdk 9 is installed too. Also, after several attempts the ubuntu .deb worked as well. Thanks for the info about running experimental (closing this issue).

closed 2018-06-17 by wotwot