axoloti/axoloti #432: Can't downgrade from experimental to master firmware.

deadsy, 2018-07-10:
Trying to revert from experimental to master firmware.

Lot's of:
SEVERE: Control transfer failed: -9
SEVERE: Control transfer failed: -4
SEVERE: Control transfer failed: -1

```
jasonh@eccles:~/work/axo$ lsb_release -a
LSB Version:    core-9.20160110ubuntu0.2-amd64:core-9.20160110ubuntu0.2-noarch:security-9.20160110ubuntu0.2-amd64:security-9.20160110ubuntu0.2-noarch
Distributor ID: Ubuntu
Description:    Ubuntu 16.04.4 LTS
Release:        16.04
Codename:       xenial
```

```
jasonh@eccles:~/work/axo$ ./run.sh 
java -jar /home/jasonh/work/axo/axoloti/dist/Axoloti.jar
defaulting to developer defaults, can be overridden
Axoloti Directories:
Current = /home/jasonh/work/axo
Jar = /home/jasonh/work/axo/axoloti/dist
Release = /home/jasonh/work/axo/axoloti
Runtime = /home/jasonh/work/axo/axoloti
Firmware = /home/jasonh/work/axo/axoloti/firmware
AxolotiHome = /home/jasonh/work/axo/axoloti
Jul 10, 2018 8:43:34 AM axoloti.utils.Preferences SavePrefs
INFO: Saving preferences...
Jul 10, 2018 8:43:34 AM axoloti.utils.Preferences SavePrefs
INFO: preferences path : /home/jasonh/work/axo/axoloti/axoloti.prefs
Jul 10, 2018 8:43:34 AM axoloti.MainFrame$4 run
INFO: Axoloti version : 1.0.12-2-0-g7975cc8-dirty  build time : 10/07/2018 08:42:04 PDT
Jul 10, 2018 8:43:34 AM axoloti.MainFrame updateLinkFirmwareID
INFO: Link to firmware CRC E95BAC96
Jul 10, 2018 8:43:34 AM axoloti.utils.AxoGitLibrary reportStatus
INFO: Status: factory (1.0.12,anon) : OK  ( 1.0.12,clean )
Jul 10, 2018 8:43:34 AM axoloti.utils.AxoFileLibrary reportStatus
INFO: Status : home : OK
Jul 10, 2018 8:43:34 AM axoloti.utils.AxoGitLibrary reportStatus
INFO: Status: community (1.0.12,anon) : OK  ( 1.0.12,dirty )
Jul 10, 2018 8:43:34 AM axoloti.utils.AxoGitLibrary reportStatus
INFO: Changes for: community (1.0.12,anon)
Jul 10, 2018 8:43:34 AM axoloti.utils.AxoGitLibrary reportStatus
INFO: modified: objects/deadsy/mpr121/mpr121.h 
Jul 10, 2018 8:43:34 AM axoloti.utils.AxoGitLibrary reportStatus
INFO: modified: objects/deadsy/ttp229/ttp229.h 
Jul 10, 2018 8:43:34 AM axoloti.utils.AxoGitLibrary reportStatus
INFO: uncommited: objects/deadsy/mpr121/mpr121.h 
Jul 10, 2018 8:43:34 AM axoloti.utils.AxoGitLibrary reportStatus
INFO: uncommited: objects/deadsy/ttp229/ttp229.h 
Jul 10, 2018 8:43:34 AM axoloti.USBBulkConnection OpenDeviceHandle
INFO: USB device found
Jul 10, 2018 8:43:34 AM axoloti.object.AxoObjects$2 run
INFO: search path : /home/jasonh/work/axo/repos/axoloti-factory/objects
Jul 10, 2018 8:43:35 AM axoloti.USBBulkConnection processByte
WARNING: file error: FR_NO_FILE, filename:"/START.BIN"
Jul 10, 2018 8:43:35 AM axoloti.USBBulkConnection connect
SEVERE: connected
Jul 10, 2018 8:43:35 AM axoloti.USBBulkConnection processByte
INFO: Firmware version: 1.1.0.0, crc=0x9BD230C9, entrypoint=0xE46D0208
Jul 10, 2018 8:43:35 AM axoloti.MainFrame setFirmwareID
SEVERE: Firmware CRC mismatch! Please flash the firmware first! Hardware firmware CRC = 9BD230C9 <> Software CRC = E95BAC96
Jul 10, 2018 8:43:35 AM axoloti.object.AxoObjects$2 run
INFO: search path : /home/jasonh/work/axo/work/objects
Jul 10, 2018 8:43:35 AM axoloti.object.AxoObjects$2 run
INFO: search path : /home/jasonh/work/axo/repos/axoloti-contrib/objects
Jul 10, 2018 8:43:37 AM axoloti.object.AxoObjects$2 run
INFO: finished loading objects
Jul 10, 2018 8:43:40 AM axoloti.USBBulkConnection connect
INFO: Authentic Axoloti Core
Jul 10, 2018 8:43:47 AM axoloti.MainFrame updateLinkFirmwareID
INFO: Link to firmware CRC E95BAC96
Jul 10, 2018 8:43:47 AM qcmds.QCmdUploadFWSDRam Do
INFO: firmware file path: /home/jasonh/work/axo/axoloti/firmware/build/axoloti.bin
Jul 10, 2018 8:43:47 AM qcmds.QCmdUploadFWSDRam Do
INFO: firmware file size: 758,504
Jul 10, 2018 8:43:47 AM qcmds.QCmdUploadFWSDRam Do
INFO: firmware crc: 0xE95BAC96
Jul 10, 2018 8:43:47 AM axoloti.USBBulkConnection UploadFragment
INFO: block uploaded @ 0xC0000000 length 16
Jul 10, 2018 8:43:47 AM qcmds.QCmdProcessor$4 run
INFO: Start uploading firmware
Jul 10, 2018 8:43:47 AM axoloti.USBBulkConnection UploadFragment
INFO: block uploaded @ 0xC0000010 length 32768
Jul 10, 2018 8:43:47 AM axoloti.USBBulkConnection UploadFragment
INFO: block uploaded @ 0xC0008010 length 32768
Jul 10, 2018 8:43:47 AM axoloti.USBBulkConnection UploadFragment
INFO: block uploaded @ 0xC0010010 length 32768
Jul 10, 2018 8:43:47 AM axoloti.USBBulkConnection UploadFragment
INFO: block uploaded @ 0xC0018010 length 32768
Jul 10, 2018 8:43:47 AM axoloti.USBBulkConnection UploadFragment
INFO: block uploaded @ 0xC0020010 length 32768
Jul 10, 2018 8:43:47 AM axoloti.USBBulkConnection UploadFragment
INFO: block uploaded @ 0xC0028010 length 32768
Jul 10, 2018 8:43:47 AM axoloti.USBBulkConnection UploadFragment
INFO: block uploaded @ 0xC0030010 length 32768
Jul 10, 2018 8:43:47 AM axoloti.USBBulkConnection UploadFragment
INFO: block uploaded @ 0xC0038010 length 32768
Jul 10, 2018 8:43:47 AM axoloti.USBBulkConnection UploadFragment
INFO: block uploaded @ 0xC0040010 length 32768
Jul 10, 2018 8:43:47 AM axoloti.USBBulkConnection UploadFragment
INFO: block uploaded @ 0xC0048010 length 32768
Jul 10, 2018 8:43:47 AM axoloti.USBBulkConnection UploadFragment
INFO: block uploaded @ 0xC0050010 length 32768
Jul 10, 2018 8:43:47 AM axoloti.USBBulkConnection UploadFragment
INFO: block uploaded @ 0xC0058010 length 32768
Jul 10, 2018 8:43:47 AM axoloti.USBBulkConnection UploadFragment
INFO: block uploaded @ 0xC0060010 length 32768
Jul 10, 2018 8:43:47 AM axoloti.USBBulkConnection UploadFragment
INFO: block uploaded @ 0xC0068010 length 32768
Jul 10, 2018 8:43:47 AM axoloti.USBBulkConnection UploadFragment
INFO: block uploaded @ 0xC0070010 length 32768
Jul 10, 2018 8:43:47 AM axoloti.USBBulkConnection UploadFragment
INFO: block uploaded @ 0xC0078010 length 32768
Jul 10, 2018 8:43:47 AM axoloti.USBBulkConnection UploadFragment
INFO: block uploaded @ 0xC0080010 length 32768
Jul 10, 2018 8:43:47 AM axoloti.USBBulkConnection UploadFragment
INFO: block uploaded @ 0xC0088010 length 32768
Jul 10, 2018 8:43:47 AM axoloti.USBBulkConnection UploadFragment
INFO: block uploaded @ 0xC0090010 length 32768
Jul 10, 2018 8:43:47 AM axoloti.USBBulkConnection UploadFragment
INFO: block uploaded @ 0xC0098010 length 32768
Jul 10, 2018 8:43:47 AM axoloti.USBBulkConnection UploadFragment
INFO: block uploaded @ 0xC00A0010 length 32768
Jul 10, 2018 8:43:48 AM axoloti.USBBulkConnection UploadFragment
INFO: block uploaded @ 0xC00A8010 length 32768
Jul 10, 2018 8:43:48 AM axoloti.USBBulkConnection UploadFragment
INFO: block uploaded @ 0xC00B0010 length 32768
Jul 10, 2018 8:43:48 AM axoloti.USBBulkConnection UploadFragment
INFO: block uploaded @ 0xC00B8010 length 4840
Jul 10, 2018 8:43:48 AM qcmds.QCmdUploadPatch Do
INFO: bin path: /home/jasonh/work/axo/axoloti/firmware/flasher/flasher_build/flasher.bin
Jul 10, 2018 8:43:48 AM axoloti.USBBulkConnection writeBytes
SEVERE: Control transfer failed: -9
Jul 10, 2018 8:43:48 AM axoloti.USBBulkConnection UploadFragment
INFO: block uploaded @ 0x20011000 length 15220
Jul 10, 2018 8:43:48 AM axoloti.USBBulkConnection writeBytes
SEVERE: Control transfer failed: -9
Jul 10, 2018 8:43:48 AM qcmds.QCmdProcessor$4 run
INFO: Done uploading firmware
Jul 10, 2018 8:43:48 AM qcmds.QCmdProcessor$4 run
INFO: Start uploading patch
Jul 10, 2018 8:43:48 AM qcmds.QCmdProcessor$4 run
INFO: Done uploading patch
Jul 10, 2018 8:43:48 AM qcmds.QCmdProcessor$4 run
INFO: Start flashing...
Jul 10, 2018 8:43:48 AM qcmds.QCmdProcessor$4 run
INFO: flashing... (ready when the green LED is steady on)
Jul 10, 2018 8:43:48 AM axoloti.USBBulkConnection writeBytes
SEVERE: Control transfer failed: -9
Jul 10, 2018 8:43:48 AM axoloti.USBBulkConnection writeBytes
SEVERE: Control transfer failed: -9
Jul 10, 2018 8:43:48 AM axoloti.USBBulkConnection writeBytes
SEVERE: Control transfer failed: -9
Jul 10, 2018 8:43:48 AM axoloti.USBBulkConnection writeBytes
SEVERE: Control transfer failed: -9
Jul 10, 2018 8:43:48 AM axoloti.USBBulkConnection writeBytes
SEVERE: Control transfer failed: -9
Jul 10, 2018 8:43:48 AM axoloti.USBBulkConnection writeBytes
SEVERE: Control transfer failed: -9
Jul 10, 2018 8:43:48 AM axoloti.USBBulkConnection writeBytes
SEVERE: Control transfer failed: -9
Jul 10, 2018 8:43:48 AM axoloti.USBBulkConnection writeBytes
SEVERE: Control transfer failed: -9
Jul 10, 2018 8:43:48 AM axoloti.USBBulkConnection writeBytes
SEVERE: Control transfer failed: -9
Jul 10, 2018 8:43:48 AM axoloti.USBBulkConnection writeBytes
SEVERE: Control transfer failed: -9
etc....
```

nicolasmiller, 2018-07-10:
Is the flash being invoked from the master or experimental patcher? I just noticed that there's an option for "Flash downgrade to 1.0.12" in the experimental patcher. Is that what you're using?

Might also be worth trying to build everything on master and use that patcher to perform a rescue which I believe requires putting the board into DFU mode.

deadsy, 2018-07-11:
I tried the downgrade command in the experimental patcher. That allowed the master patcher to load the firmware correctly. From this I take it that the behavior is recognized and intentional. It's inconvenient and it's likely to catch out others, but I assume there is a reason for the break in forward/backward compatibility of firmware loading. As such I'll close the issue.

closed 2018-07-11 by deadsy