axoloti/axoloti #437: Please add HTTPS to http://community.axoloti.com

absorber, 2018-09-30:
Hello,

I feel uncomfortable browsing and posting on the forum over plaintext HTTP, especially since it requires Javascript. There are quite a few risks for not using HTTPS, see [here](https://www.troyhunt.com/heres-why-your-static-website-needs-https/) for example.

It's pretty easy and free these days, especially using [Let's Encrypt](https://letsencrypt.org/). There are numerous guides out there, for example: https://www.troyhunt.com/https-is-easy/

There's hardly ever a reason not to have HTTPS in 2018.

Thanks in advance.

ramonschepers, 2018-09-30:
i second this one. asked it before, but it wasn't top priority back then.

CarloCattano, 2021-01-10:
+1 ,is 2021 already

CarloCattano, 2022-06-22:
Is the forum down for good ?

chkworks, 2023-06-02:
Time to move on...