axoloti/axoloti #438: error when importing std::complex

deepinthewoods, 2018-10-21:
this error happens with just an empty object with `#include <complex>` in the Local Data

`
Generate code complete
Start compiling patch
BDIR = C:\Users\Niall\DOCUME~1\axoloti/build
FIRMWARE = .
RM
APP
! C:\Users\Niall\DOCUME~1\axoloti/build/xpatch.h.gch
. c:\program files (x86)\axoloti_runtime\platform_win\arm-none-eabi\include\c++\4.9.3\complex
.. c:\program files (x86)\axoloti_runtime\platform_win\arm-none-eabi\include\c++\4.9.3\arm-none-eabi\armv7e-m\fpu\bits\c++config.h
.. c:\program files (x86)\axoloti_runtime\platform_win\arm-none-eabi\include\c++\4.9.3\bits\cpp_type_traits.h
.. c:\program files (x86)\axoloti_runtime\platform_win\arm-none-eabi\include\c++\4.9.3\ext\type_traits.h
.. c:\program files (x86)\axoloti_runtime\platform_win\arm-none-eabi\include\c++\4.9.3\cmath
... c:\program files (x86)\axoloti_runtime\platform_win\arm-none-eabi\include\math.h
.. c:\program files (x86)\axoloti_runtime\platform_win\arm-none-eabi\include\c++\4.9.3\sstream
... c:\program files (x86)\axoloti_runtime\platform_win\arm-none-eabi\include\c++\4.9.3\istream
.... c:\program files (x86)\axoloti_runtime\platform_win\arm-none-eabi\include\c++\4.9.3\ios
..... c:\program files (x86)\axoloti_runtime\platform_win\arm-none-eabi\include\c++\4.9.3\iosfwd
..... c:\program files (x86)\axoloti_runtime\platform_win\arm-none-eabi\include\c++\4.9.3\exception
..... c:\program files (x86)\axoloti_runtime\platform_win\arm-none-eabi\include\c++\4.9.3\bits\char_traits.h
..... c:\program files (x86)\axoloti_runtime\platform_win\arm-none-eabi\include\c++\4.9.3\bits\localefwd.h
..... c:\program files (x86)\axoloti_runtime\platform_win\arm-none-eabi\include\c++\4.9.3\bits\ios_base.h
...... c:\program files (x86)\axoloti_runtime\platform_win\arm-none-eabi\include\c++\4.9.3\ext\atomicity.h
...... c:\program files (x86)\axoloti_runtime\platform_win\arm-none-eabi\include\c++\4.9.3\bits\locale_classes.h
....... c:\program files (x86)\axoloti_runtime\platform_win\arm-none-eabi\include\c++\4.9.3\string
....... c:\program files (x86)\axoloti_runtime\platform_win\arm-none-eabi\include\c++\4.9.3\bits\locale_classes.tcc
..... c:\program files (x86)\axoloti_runtime\platform_win\arm-none-eabi\include\c++\4.9.3\streambuf
...... c:\program files (x86)\axoloti_runtime\platform_win\arm-none-eabi\include\c++\4.9.3\bits\streambuf.tcc
..... c:\program files (x86)\axoloti_runtime\platform_win\arm-none-eabi\include\c++\4.9.3\bits\basic_ios.h
...... c:\program files (x86)\axoloti_runtime\platform_win\arm-none-eabi\include\c++\4.9.3\bits\locale_facets.h
....... c:\program files (x86)\axoloti_runtime\platform_win\arm-none-eabi\include\c++\4.9.3\cwctype
........ c:\program files (x86)\axoloti_runtime\platform_win\arm-none-eabi\include\wctype.h
......... c:\program files (x86)\axoloti_runtime\platform_win\arm-none-eabi\include\_ansi.h
......... c:\program files (x86)\axoloti_runtime\platform_win\arm-none-eabi\include\sys\_types.h
......... c:\program files (x86)\axoloti_runtime\platform_win\lib\gcc\arm-none-eabi\4.9.3\include\stddef.h
....... c:\program files (x86)\axoloti_runtime\platform_win\arm-none-eabi\include\c++\4.9.3\cctype
........ c:\program files (x86)\axoloti_runtime\platform_win\arm-none-eabi\include\ctype.h
....... c:\program files (x86)\axoloti_runtime\platform_win\arm-none-eabi\include\c++\4.9.3\arm-none-eabi\armv7e-m\fpu\bits\ctype_base.h
....... c:\program files (x86)\axoloti_runtime\platform_win\arm-none-eabi\include\c++\4.9.3\ext\numeric_traits.h
....... c:\program files (x86)\axoloti_runtime\platform_win\arm-none-eabi\include\c++\4.9.3\bits\streambuf_iterator.h
........ c:\program files (x86)\axoloti_runtime\platform_win\arm-none-eabi\include\c++\4.9.3\debug\debug.h
....... c:\program files (x86)\axoloti_runtime\platform_win\arm-none-eabi\include\c++\4.9.3\arm-none-eabi\armv7e-m\fpu\bits\ctype_inline.h
....... c:\program files (x86)\axoloti_runtime\platform_win\arm-none-eabi\include\c++\4.9.3\bits\locale_facets.tcc
...... c:\program files (x86)\axoloti_runtime\platform_win\arm-none-eabi\include\c++\4.9.3\bits\basic_ios.tcc
.... c:\program files (x86)\axoloti_runtime\platform_win\arm-none-eabi\include\c++\4.9.3\ostream
..... c:\program files (x86)\axoloti_runtime\platform_win\arm-none-eabi\include\c++\4.9.3\bits\ostream_insert.h
..... c:\program files (x86)\axoloti_runtime\platform_win\arm-none-eabi\include\c++\4.9.3\bits\ostream.tcc
...... c:\program files (x86)\axoloti_runtime\platform_win\arm-none-eabi\include\c++\4.9.3\bits\cxxabi_forced.h
.... c:\program files (x86)\axoloti_runtime\platform_win\arm-none-eabi\include\c++\4.9.3\bits\istream.tcc
... c:\program files (x86)\axoloti_runtime\platform_win\arm-none-eabi\include\c++\4.9.3\bits\sstream.tcc
In file included from c:\program files (x86)\axoloti_runtime\platform_win\arm-none-eabi\include\c++\4.9.3\bits\ios_base.h:41:0,
                 from c:\program files (x86)\axoloti_runtime\platform_win\arm-none-eabi\include\c++\4.9.3\ios:42,
                 from c:\program files (x86)\axoloti_runtime\platform_win\arm-none-eabi\include\c++\4.9.3\istream:38,
                 from c:\program files (x86)\axoloti_runtime\platform_win\arm-none-eabi\include\c++\4.9.3\sstream:38,
                 from c:\program files (x86)\axoloti_runtime\platform_win\arm-none-eabi\include\c++\4.9.3\complex:45,
                 from C:\Users\Niall\DOCUME~1\axoloti/build/xpatch.cpp:40:
c:\program files (x86)\axoloti_runtime\platform_win\arm-none-eabi\include\c++\4.9.3\bits\locale_classes.h:43:1: error: expected unqualified-id before 'namespace'
 namespace std _GLIBCXX_VISIBILITY(default)
 ^
C:\Users\Niall\DOCUME~1\axoloti/build/xpatch.cpp:513:1: error: expected '}' at end of input
 }
 ^
C:\Users\Niall\DOCUME~1\axoloti/build/xpatch.cpp:513:1: error: expected unqualified-id at end of input
C:\Users\Niall\DOCUME~1\axoloti/build/xpatch.cpp:513:1: error: expected '}' at end of input
C:\Users\Niall\DOCUME~1\axoloti/build/xpatch.cpp:513:1: error: expected unqualified-id at end of input
Multiple include guards may be useful for:
c:\program files (x86)\axoloti_runtime\platform_win\arm-none-eabi\include\c++\4.9.3\arm-none-eabi\armv7e-m\fpu\bits\ctype_base.h
c:\program files (x86)\axoloti_runtime\platform_win\arm-none-eabi\include\c++\4.9.3\arm-none-eabi\armv7e-m\fpu\bits\ctype_inline.h
c:\program files (x86)\axoloti_runtime\platform_win\arm-none-eabi\include\c++\4.9.3\cctype
c:\program files (x86)\axoloti_runtime\platform_win\arm-none-eabi\include\c++\4.9.3\cmath
c:\program files (x86)\axoloti_runtime\platform_win\arm-none-eabi\include\c++\4.9.3\cwctype
c:\program files (x86)\axoloti_runtime\platform_win\lib\gcc\arm-none-eabi\4.9.3\include\stddef.h
make: *** [C:\Users\Niall\DOCUME~1\axoloti/build/xpatch.bin] Error 1
shell task failed, exit value: 1
Compiling patch failed ( untitled ) `

TheTechnobear, 2018-10-21:
works here 
(mac os, but its the same compiler so can't imagine being different) 

how are you including it?

you should not be doing #include in the object, you need to edit the object and place it in the includes section.

the following works fine.
`<objdefs appVersion="1.0.12">
   <obj.normal id="inctest" uuid="2752418a-4864-444d-b6b9-917e0c078808">
      <license>GPL</license>
      <inlets/>
      <outlets/>
      <displays/>
      <params>
      </params>
      <attribs/>
      <includes>
         <include>complex</include>
      </includes>
      <code.declaration></code.declaration>
      <code.krate></code.krate>
   </obj.normal>
</objdefs>`

TheTechnobear, 2019-12-20:
also cannot reproduce in 2.0, lets close - reopen if necessary.

closed 2019-12-20 by TheTechnobear