axoloti/axoloti #448: platform_osx/paths.sh

TheTechnobear, 2019-12-19:
axoloti 2.0.0

hard coded to PATH=/Applications/gcc-arm-none-eabi-7-2018-q2-update/bin

I dont think /Applications is a good place on macOS , /usr/local is more appropriate for dev tools.

also I dont think we should hard code version, if we are going to allow for different (more up to date version) 

my suggestion is we change above to something like
/usr/local/axoloti-gcc-arm-none-eabi 

and then get users to create a symbolic link to the specific version they use
e.g. 
/usr/local/bin/axoloti-gcc-arm-none-eabi -> /usr/local/gcc-arm-none-eabi-9-2019-q4-major

then use PATH=/usr/local/axoloti-gcc-arm-none-eabi /bin

TheTechnobear, 2019-12-19:
note: im not sure if this is only relevant to macOS, of if windows/linux needs something similar?
(Id assume linux, wont have an issue as the path is managed by apt ?! )

rekado, 2019-12-26:
>  (Id assume linux, wont have an issue as the path is managed by apt ?! )

`apt` is only used by Debian and its derivatives.  It doesn't manage the PATH variable.

For the `axoloti-patcher-next` package in GNU Guix I ignored the hard-coded PATH and instead patched `env/arm-none-eabi.mk` to refer to the absolute file name of the toolchain (which is what we usually do on Guix anyway).  This means that instead of looking up arm-none-eabi-* on PATH we're invoking the specific tools of the toolchain by their absolute names.

Generally, I think that setting PATH is a kludge and should be avoided.

TheTechnobear, 2019-12-26:
hardcoding paths is generally a bad idea. 

Dev usually use something like ./configure to find appropriate tools.   
Though if we have a large number of dependencies something like a axoloti-tools package could be useful, where dependencies get installed.  

( axoloti-tools can then be reused for users, as below)

For users, package management allows for dependencies to be specified. That’s the way i do it for Debian and arch Linux. 
If there are not deb repo or aur for some packages we could easily create one. 

Id have thought Debian and arch Linux will cover most users.

TheTechnobear, 2019-12-26:
Generally I think if Linux is a popular user platform - we should setup a repo for arch Linux and Debian for end users -  it’s pretty simple. 

That said , I’ve still seen very few Linux uses in the forum - and those that there are all seem to require something different - so not sure if the maintenance effort is justified.

rekado, 2019-12-26:
> Though if we have a large number of dependencies something like a axoloti-tools package could be useful, where dependencies get installed.

As a package maintainer I'm frustrated with programs that try to use the system package manager to install packages.  Instead the application itself should be packaged for the system package manager and use its facilities for specifying dependencies.

(Axoloti *does* have a non-trivial number of dependencies, including patched variants of other libraries and tools.  Here's the Guix package definition for reference: https://git.savannah.gnu.org/cgit/guix.git/tree/gnu/packages/axoloti.scm#n362)

Using conventional build systems makes this process much easier for package maintainers of GNU/Linux distributions.

Luckily much of Axoloti can be built with the conventional ant build system (via build.xml), so throwing in another build system (such as a configure script used by the GNU build system) would be a little weird.  I suppose it would be sufficient to make the location of the cross toolchain configurable via defines / properties in build.xml.

> Generally I think if Linux is a popular user platform - we should setup a repo for arch Linux and Debian for end users - it’s pretty simple.

FWIW, the Guix package can be installed on *any* GNU/Linux distribution, including Debian.

TheTechnobear, 2019-12-26:
‘’’
FWIW, the Guix package can be installed on any GNU/Linux distribution, including Debian.
‘’’

What’s the advantage of install another package management system on distros that already have perfectly good package management systems ?
Seems like extra hassle to end users to me ?!

zrna-research, 2019-12-26:
I agree with @rekado here. I think it's a mistake to bother with distro-specific packages, especially when the Linux userbase is smaller. I would suggest going even further to something like https://appimage.org/ that doesn't even require the user to bother with a package manager at all.

Removing the ARM toolchain binary from the package seems like a mistake on Johannes' part. We shouldn't be forcing the average end user to think about that. We should include the exact known to be working binaries like we were doing before. The style of having build scripts or git fetch stuff like the ARM toolchain and ChibiOS and even the factory objects I think is a mistake as well. We should be able to create a single archive that has everything and doesn't need to fetch things that might ultimately be unavailable when URLs change or servers go down. This has bitten us repeatedly. The package should be standalone. 

In that case a user would just have version x.y.z of the patcher, firmware and factory objects all in a single standalone package, generated from a monolithic repository that stores everything necessary in order to run, leaving nothing dangling at remote git repos or URLs or isolated runtimes, etc.

In this case the user just manages upgrades as they see fit, i.e. they have an isolated directory at version 1.x.y and then they create a new directory with version 1.x.(y+1) and so on and there's no possibility of breakage or confusion. They can easily run multiple versions. They can roll back to a previous version easily. They can upgrade to a new release slowly without trashing work on a previous stable release.