axoloti/axoloti #449: detailed instructions for axoloti install (2.0.0)

TheTechnobear, 2019-12-19:
(note: ive only installed macOS at this point, so not sure relevance of this to windows/linux yet) 

we need a bit of clarity on installation of gcc-arm-none-eabi

whilst developers im sure wont have issues, i think its going to be a bit daunting for some end-users - they are used to a kind of one click install.

I think we need to be 'prescriptive' over where to install it, and then default the additional path in axoloti preferences. also are we really happy that 'any' version can be used? 

I think we need to consider two concerns:
a) support
having users on different versions of gcc-arm-none-eabi, could potentially lead to users having different experiences ... and be a nightmare to support.

b) end users
we dont want to raise the 'skill' level required to use axoloti, one of its attractions is for its a graphic environment, you dont need to know c++, command line, compilers.

(of course, its great that advanced users can then change the defaults as they wish, e.g. have different/newer gcc-arm-none-eabi versions, or locate where they want) 

note: I do really like the fact that we dont have the old axoloti runtime!
definitely a good move forward, so this is more about a 'refinement' of the new way :)

rekado, 2019-12-20:
The Axoloti patcher can be installed with GNU Guix on all GNU/Linux systems with the command `guix install axoloti-patcher`.  This includes the arm-none-eabi cross compiler.

For users that don't have Guix we can provide self-contained binary bundles containing all dependencies (using `guix pack`). 

I'm currently working on a package for version 2.0.0.

Pastitas, 2020-04-11:
> The Axoloti patcher can be installed with GNU Guix on all GNU/Linux systems with the command `guix install axoloti-patcher`. This includes the arm-none-eabi cross compiler.
>
> For users that don't [...]

Would it be possible for you to attach the binary bundles for 1.0.12?

Deftaudio, 2020-12-26:
It's been a long time since I installed Axoloti. I've just moved to another Mac (High Sierra) and totally lost with Axoloti 2.0 install. Can you please clarify how properly install it and set up paths to arm-none-eabi-strip? I installed gcc-arm-none-eabi-10-2020-q4-major-mac.pkg but seems it didn't help...

3Dietrich, 2022-10-21:
The same as @Deftaudio with me: I have an Idea to use axoloti, but need to re-install it and every time I want to set it live it ends in errors: 
`/bin/sh: arm-none-eabi-g++: command not found
 ... Error 127 Compiling patch failed`
what is "arm-none-eabi-g++? How to installe it on mac M1
`guix` is totally new to me and also does not work. I found 
`sudo port install arm-none-eabi-gcc` but this gives me only errors.. and this is also not "arm-none-eabi-g++".. HELP!

rekado, 2022-10-21:
`arm-none-eabi-g++` is a cross-compiler, a variant of GCC built with newlib-nano (instead of the GNU C Library) that runs on your host machine and produces executables that run on the axoloti board.
Guix is not available on macos.

3Dietrich, 2022-10-21:
thank you @rekado !
I' m still not getting axoloti to run "Live". get allways the same error:
`Generate code complete
compiling /Users/dpa/Documents/axoloti-2.0.0/build/xpatch.cpp
/bin/sh: arm-none-eabi-g++: command not found
make: *** [/Users/dpa/Documents/axoloti-2.0.0/build/xpatch.o] Error 127
Compiling patch failed`
what is going wrong here?

rekado, 2022-10-21:
That means that you don't have the cross compiler executable.

3Dietrich, 2022-10-21:
I mean.. I found the files for this in
`/Applications/axoloti_runtime/platform_osx/bin`
I tried to move the folder but i did not found any possibility to transmit "Axoloti! look here catch it!

rekado, 2022-10-21:
The file axoloti.prefs should have a reference to the runtime directory.  If the value for RuntimeDir is in fact consistent with the directory containing the complier then it would be worth checking if the cross-compiler is actually executable on your architecture.

Can you invoke the arm-none-eabi-g++ command and see if this prints an error?  My guess is that the binary is built for the wrong architecture and won't run on your M1 CPU.

3Dietrich, 2022-10-21:
thank you @rekado but the axoloti.prefs only have the Libraries from the preference. it doesn't contain anything similar to "runtime".

Actually in the axioloti console was an instruction I oversaw. I followed the instructions now (downlodaded a folder "gcc-arm-none-eabi-7-2018-q2-update" and placed this in Applications/) and after restarting and **a lot** of safety popups and agreements it finally worked! All other paths I tested did not work. Now I have a folder 'gcc-arm-none-eabi-7-2018-q2-update' in the Application folder. It feels a bit disordered but hey, axioloti works!

sorry that I abused this thread for my problems. But maybe someone else has the same problem and asks for a better install instruction :)
My tip: read the console carefully, not just only the red. Even if it feels bad (e.g. copying an order named "gcc-arm-none-eabi-7-2018-q2-update" into Applications/) follow it.