axoloti/axoloti #450: writing midi settings results in exceptions and errors

TheTechnobear, 2019-12-20:
steps
board->midi routing
refresh
write settings

leads to following errors.
also it appears the routing is not working on the board.

as toggle->midi/out/clock run leads to midi overflow  (as no device connected?) 

[code]
uploading: /settings/midi-out/DIN.axr
Control transfer failed: -1
receive error: LIBUSB_ERROR_IO
Disconnect request
java.io.IOException
java.io.IOException
	at axoloti.connection.USBBulkConnection_v2.createDirectory(USBBulkConnection_v2.java:1012)
	at axoloti.connection.USBBulkConnection_v2.createDirIfNonExistant(USBBulkConnection_v2.java:579)
	at axoloti.connection.USBBulkConnection_v2.upload(USBBulkConnection_v2.java:598)
	at axoloti.target.midirouting.MidiOutputRoutingTable.lambda$upload$0(MidiOutputRoutingTable.java:136)
	at axoloti.job.JobProcessor.lambda$exec$0(JobProcessor.java:21)
	at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1149)
	at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:624)
	at java.lang.Thread.run(Thread.java:748)

uploading: /settings/midi-out/USB device.axr
java.io.IOException: Not connected
java.io.IOException: Not connected
	at axoloti.connection.USBBulkConnection_v2.transmitPacket1(USBBulkConnection_v2.java:415)
	at axoloti.connection.USBBulkConnection_v2.getFileInfo(USBBulkConnection_v2.java:491)
	at axoloti.connection.USBBulkConnection_v2.createDirIfNonExistant(USBBulkConnection_v2.java:577)
	at axoloti.connection.USBBulkConnection_v2.upload(USBBulkConnection_v2.java:598)
	at axoloti.target.midirouting.MidiOutputRoutingTable.lambda$upload$0(MidiOutputRoutingTable.java:136)
	at axoloti.job.JobProcessor.lambda$exec$0(JobProcessor.java:21)
	at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1149)
	at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:624)
	at java.lang.Thread.run(Thread.java:748)

uploading: /settings/midi-in/DIN.axr
java.io.IOException: Not connected
java.io.IOException: Not connected
	at axoloti.connection.USBBulkConnection_v2.transmitPacket1(USBBulkConnection_v2.java:415)
	at axoloti.connection.USBBulkConnection_v2.getFileInfo(USBBulkConnection_v2.java:491)
	at axoloti.connection.USBBulkConnection_v2.createDirIfNonExistant(USBBulkConnection_v2.java:577)
	at axoloti.connection.USBBulkConnection_v2.upload(USBBulkConnection_v2.java:598)
	at axoloti.target.midirouting.MidiInputRoutingTable.lambda$upload$0(MidiInputRoutingTable.java:81)
	at axoloti.job.JobProcessor.lambda$exec$0(JobProcessor.java:21)
	at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1149)
	at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:624)
	at java.lang.Thread.run(Thread.java:748)

uploading: /settings/midi-in/USB device.axr
java.io.IOException: Not connected
java.io.IOException: Not connected
	at axoloti.connection.USBBulkConnection_v2.transmitPacket1(USBBulkConnection_v2.java:415)
	at axoloti.connection.USBBulkConnection_v2.getFileInfo(USBBulkConnection_v2.java:491)
	at axoloti.connection.USBBulkConnection_v2.createDirIfNonExistant(USBBulkConnection_v2.java:577)
	at axoloti.connection.USBBulkConnection_v2.upload(USBBulkConnection_v2.java:598)
	at axoloti.target.midirouting.MidiInputRoutingTable.lambda$upload$0(MidiInputRoutingTable.java:81)
	at axoloti.job.JobProcessor.lambda$exec$0(JobProcessor.java:21)
	at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1149)
	at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:624)
	at java.lang.Thread.run(Thread.java:748)

USB device found
connected
Authentic Axoloti Core
Firmware version: 2.0.0.0, crc=0xCDEEEFDA
exception report:
pc=0x2FE528
psr=0x21000000
lr=0x801ABBF
r12=0x0
cfsr=0x100
[/code[

TheTechnobear, 2019-12-20:
note: this is 2.0.0

also prior to this board is connected, and is able to run patches...
so its unclear why the get the control transfer failed - which leads to disconnection and then ultimately other errors.

TheTechnobear, 2019-12-20:
ok, the issue is 
[code]
at axoloti.connection.USBBulkConnection_v2.createDirectory(USBBulkConnection_v2.java:1012)
at axoloti.connection.USBBulkConnection_v2.createDirIfNonExistant(USBBulkConnection_v2.java:579)
[/code]

whilst its attempting to create the midi-out directory, it does not create the settings directory

workaround: create the settings directory manually on the card, then it will work

fix: check for settings directory, and create if necessary.

JohannesTaelman, 2020-01-24:
Can't reproduce this issue, after deleting the settings directory, it is recreated when I do: board->midi routing, refresh, write settings...
Missing parent directories are created in USBConnectionv2.upload(). 
The real issue is perhaps something else than missing parent directories.