axoloti/axoloti #453: factory objects - not working

TheTechnobear, 2019-12-21:
a list of objects that need revision for 2.0.0 

please add other modules you find in comments below.
(if appropriate say why .. is it not working, not compiling)

note: once majority of these are done, this issue can be closed, and we can create issue or specific objects.

mutable: (need to be placed in modules)
fx/lmnts/lmnts
fx/lmnts/string
fx/lmnts/tube
fx/clds/clds
fx/clds/pitchshifter
fx/rngs/chorus
fx/rngs/ensemble
fx/rngs/reverb
fx/strms/strms
fx/wrps/wrps

utils/format - not compile - due to new api, to use sd card, does not compile
osc/brds/csaw (?)

acik1, 2019-12-28:
math/log and math/exp don't work. 
Console output:

Done compiling patch
Patch failed to load : failed to load patch
Can't find address for symbol logt (1)

mlinton, 2020-01-09:
add the jaffa/filt/ladder1 module to the list as well