axoloti/axoloti #455: Axoloti core crashes on 2.0.0

janifr, 2020-01-04:
Hi,

I came up with a axo crash and reboot. In the zip there are patch that crashes and two slightly modified ones that do not.

Before I got in to crashing state, I was hunting for a strange bug I couldn't deal with. Something was writing strange values to array click_velocity[], even I commented out every line referencing that array.

Here's a log from the main window. Any way I can help with this thing sorted out?

`Link to firmware CRC CDEEEFDA
Axoloti version : 2.0.0-0-g88b60feb  build time : 18/12/2019 01:10:50 UTC
Status: factory (2.0.0,anon) : OK  ( 2.0.0,clean )
Status : home : OK
Status: community (2.0.0,anon) : OK  ( 2.0.0,clean )
USB device found
connected
Authentic Axoloti Core
search path : /home/jani/axoloti-2.0.0/axoloti-factory/objects
Firmware version: 2.0.0.0, crc=0xCDEEEFDA
search path : /home/jani/axoloti-2.0.0/objects
search path : /home/jani/axoloti-2.0.0/axoloti-contrib/objects
finished loading objects
Generate code complete
compiling /home/jani/axoloti-2.0.0/build/xpatch.cpp
linking /home/jani/axoloti-2.0.0/build/xpatch.dbg.elf
/home/jani/axoloti-2.0.0/build/xpatch.elf  :
section            size   addr
.text              8432      0
.rodata            2960      0
.bss                  8      0
.init_array           0      0
.fini_array           0      0
.comment            442      0
.ARM.attributes      57      0
Total             11899

Done compiling patch
receive error: LIBUSB_ERROR_IO
Control transfer failed: -1
Disconnect request
java.io.IOException: Not connected
java.io.IOException: Not connected
	at axoloti.connection.USBBulkConnection_v2.transmitPacket1(USBBulkConnection_v2.java:415)
	at axoloti.connection.USBBulkConnection_v2.patchGetName(USBBulkConnection_v2.java:744)
	at axoloti.connection.USBBulkConnection_v2.lambda$process_pckt_patch_list$0(USBBulkConnection_v2.java:1359)
	at java.lang.Thread.run(Thread.java:748)

USB device found
connected
Authentic Axoloti Core
Firmware version: 2.0.0.0, crc=0xCDEEEFDA
exception report:
pc=0x0
psr=0x0
lr=0x5E3
r12=0x5F737361
cfsr=0x20000
`

[drawbar_crash.zip](https://github.com/axoloti/axoloti/files/4022294/drawbar_crash.zip)

zrnsm, 2020-01-04:
Few quick thoughts:

- If you take a look at the STM32F427 datasheet's definition of the CFSR register, you'll see that 0x20000 indicates `INVSTATE` which means:
"1: The processor has attempted to execute an instruction that makes illegal use of the
EPSR"
My gut feeling without step debugging is that somehow the failure has to do with ensuring that storage is available for `click_velocity` and that that name can be resolved with the new 2.0 linking approach.
- What strikes me as weird about this is that it seems like `int16_t keys[nkeys];` is declared and accessed without issue. `int16_t velocity[200];` and `int16_t click_velocity[200];` are similarly declared. One distinction is that `keys` is zero initialized:
`int i;
for(i=0;i<nkeys;i++) {
	keys[i]=0;
}
`
It seems like `click_velocity` is never zero initialized.
- I'd try to whittle it down to the simplest possible example that fails; it seems like the first one is a simpler delta relative to the one that doesn't crash (_click vs _click_crash). It may be as simple as `i` going outside the bounds of the array:
`if (attr_input.click_velocity[i]&&in1)`
Try to figure out exactly what the value of `i` is when the crash occurs or if there's some simpler type of expression involving `click_velocity` that will fail.

janifr, 2020-01-05:
Thanks for pointing this out. I was referencing outside of array velocity[200]. This issue had nothing to with axoloti.

This reminds me the importance of writing easily understandable code. Need to rewrite some modules completely.

closed 2020-01-05 by janifr