axoloti/axoloti #456: Space substitution in Makefile.patch not working

osch, 2020-03-27:
With Axoloti Patcher version 1.0.12-2 and using GNU Make 4.3 I examined the following problem:

When compiling a patch I'm getting the error:

```
Start compiling patch
make: *** No rule to make target '/home/oliver/.local/share/axoloti/build /xpatch.cpp', needed by '/home/oliver/.local/share/axoloti/build /xpatch.bin'.  Stop.

```

I tracked it down to file `firmware/Makefile.patch`  line 24 - 26:

```Makefile
SPACE := 
SPACE += 
BDIR=$(subst $(SPACE),\ ,${axoloti_home}/build)
```

However this does not work with GNU Make 4.3. Take the following example Makefile:

```Makefile
x := abc
SPACE := 
SPACE += 
y = $(subst $(SPACE),\ ,$(x))

default:
	@echo "_$(y)_"
```

Above Makefile gives the output:
```
_abc\ _
```

mongrol, 2020-06-13:
Bug and fix confirmed on Arch Linux.

elatomo, 2021-04-11:
I can also confirm the bug and the fix in GNU Make 4.3.

I'm no Make guru, but the proposed solution by @osch seems to be the appropiate one. From the [docs](https://ftp.gnu.org/old-gnu/Manuals/make-3.79.1/html_chapter/make_6.html):

> Whitespace around the variable name and immediately after the `=' is ignored.

@osch, what do you think about opening a PR with your fix?

osch, 2021-04-11:
My above observations are refering to stable version 1.0.12-2. However there is no branch for the 1.0.X releases. The situation seems to have changed on current master branch, at least the file `firmware/Makefile.patch` has vanished. So I don't know, against which branch I should open a pull request.

elatomo, 2021-04-11:
You're totally right, my wrong. I also was checking 1.0.12-2 and didn't notice that the file is no longer available in master. Seems it was removed in 4e6644225bbd61cdcedcd5f50061da209bda163f :man_shrugging: