axoloti/axoloti #461: PA2 and PA3 analog GPIO doesn't work on 2.0.0

andras, 2020-07-29:
Regression from version 1: PA2 and PA3 (both driver by a potentiometer on the AxoControl) both read noise on 2.0.0.  On version 1, they read the pot voltages.

andras, 2020-12-09:
The last release was a year ago, and there doesn't appear to be much work going on on release 2.  Is this bug ever going to be fixed by someone who actually knows where to start looking in the code?

nikristovski, 2020-12-29:
Hi, just wanted to post here and say I'm experiencing the exact same issue with my Axoloti. The analog inputs PA2 and PA3 are just using the values from PA1 (but much more unstable, as if these inputs are floating). I can't confirm if they worked on previous firmwares though...